silveirago/DIY-Midi-Controller #13: potCState overwritten

harbars, 2023-11-17:
Lines 245 & 247:    

potCState[i] = responsivePot[i].getValue();

potCState[i] = analogRead(POT_ARDUINO_PIN[i]);  // reads the pins from arduino

silveirago, 2023-11-18:
You need to instal the ResponsiveAnalogRead library

harbars, 2023-11-18:
yes, I know that. 

the problem here is the first line *uses* RAR and places it's value into potCState[i], and then you immediately discard that value by calling analogRead() in the next line and reading directly from the analog input. 

Line 247 is unnecessary, and means the RAR calculated value will never be used in this example

silveirago, 2023-11-25:
You are absolutely right. When I added the ResponsiveAnalogRead I forgot to delete that. Thanks!
The code is updated now.

closed 2023-11-25 by silveirago